mtongnz/ESP8266_ArtNetNode_v2 #144: I have no clue what I am doing :(

D0rsty, 2022-12-12:
Hi, I find it very hard to follow written instructions and learn better visually.  Came across this node idea and thought it would be amazing for what I do in theatres.  If anyone would be willing to make a tutorial of sorts/or  a video for me I would be eternally grateful.

DrStein99, 2023-02-01:
https://www.instructables.com/ESP8266-Artnet-to-DMX/

So yea, I need help paying my mortgage and want a Lamborghini in my driveway.  If you can help out with that, would be just great.  Until then, your welcome to the link for the tutorial that comes up on a basic google search.

LoneWalkerWolf, 2023-04-22:
Feel free to email me at [alexander@expanseelectronics.com](mailto:alexander@expanseelectronics.com). I have unofficially taken over this project, as Matt seems to be offline / not contributing to this project anymore. We sell a version of this project on our website: [expanseelectronics.com](https://expanseelectronics.com), and you are free to send us an email at any time with any further questions. 

We have some basic documentation available by email, and we'd be happy to send them over to you.

casesolved-co-uk, 2024-01-18:
https://github.com/mtongnz/ESP8266_ArtNetNode_v2/issues/148