mtongnz/ESP8266_ArtNetNode_v2 #145: schematic for wemos d1 mini?

lesleylemmens, 2023-04-01:
hi,
is there a schematic for wemos d1 mini?
A fritzing schematic style would the best.
I can only find a schematic for a totaal dit 8266 solution

LoneWalkerWolf, 2023-04-22:
Feel free to email me at [alexander@expanseelectronics.com](mailto:alexander@expanseelectronics.com). I can definitely send you a schematic for a D1 mini. 

In response to your other questions: 
- The module you've linked, the side with `DI DE RE RO`, is the side for the ESP. The other side, which is labelled `GND A B VCC` is the side for DMX. 
- In the schematic you've attached, the pixels are driven using the IO2 (for port B) and TX (for port 1). Be careful not to enable DMX output on these pins, however, as this could damage your WS2812B LEDs. I'd recommend using the `convert_max485_to_pix.png` schematic linked [here](https://github.com/mtongnz/ESP8266_ArtNetNode_v2/blob/master/schematics/convert_max485_to_pix.png) as an extra safety feature. 
- No, it is not possible to receive DMX and then convert it to pixels. You can only use it to output pixel data. Can I suggest you look at the WLED project for that? Linked [here](https://github.com/Aircoookie/WLED). 

I have unofficially taken over this project, as Matt seems to be offline / not contributing to this project anymore. We sell a version of this project on our website: [expanseelectronics.com](https://expanseelectronics.com), and you are free to send us an email at any time with any further questions.

lesleylemmens, 2023-04-23:
Thank you for your response!

finnie2006, 2023-09-11:
> Feel free to email me at [alexander@expanseelectronics.com](mailto:alexander@expanseelectronics.com). I can definitely send you a schematic for a D1 mini.
>
> In response to your other questions: [...]

You have every right to do this, but why are you offering an open source project for sale on your own hardware for a pretty high amount of money

casesolved-co-uk, 2024-01-18:
https://github.com/mtongnz/ESP8266_ArtNetNode_v2/issues/148