mtongnz/ESP8266_ArtNetNode_v2 #146: ESP8266_ArtNetNode_v2

ozdemirmcihad, 2023-05-20:
I downloaded the entire project, I downloaded the necessary libraries, but I'm getting a lot of errors. Is there an error in the project file?

some of the mistakes.

C:\Users\ozdem\Desktop\ESP8266_ArtNetNode_v2-master\source\espArtnetNode_2.0.0_b5g\espArtnetNode_2.0.0_b5g.ino: In function 'void setup()':
C:\Users\ozdem\Desktop\ESP8266_ArtNetNode_v2-master\source\espArtnetNode_2.0.0_b5g\espArtnetNode_2.0.0_b5g.ino:221:3: error: 'wifiStart' was not declared in this scope; did you mean 'wifiStatus'?
  221 |   wifiStart();
      |   ^~~~~~~~~
      |   wifiStatus
C:\Users\ozdem\Desktop\ESP8266_ArtNetNode_v2-master\source\espArtnetNode_2.0.0_b5g\espArtnetNode_2.0.0_b5g.ino:224:3: error: 'webStart' was not declared in this scope
  224 |   webStart();
      |   ^~~~~~~~
C:\Users\ozdem\Desktop\ESP8266_ArtNetNode_v2-master\source\espArtnetNode_2.0.0_b5g\espArtnetNode_2.0.0_b5g.ino:235:5: error: 'artStart' was not declared in this scope
  235 |     artStart();
      |     ^~~~~~~~
C:\Users\ozdem\Desktop\ESP8266_ArtNetNode_v2-master\source\espArtnetNode_2.0.0_b5g\espArtnetNode_2.0.0_b5g.ino:242:5: error: 'portSetup' was not declared in this scope
  242 |     portSetup();
      |     ^~~~~~~~~
C:\Users\ozdem\Desktop\ESP8266_ArtNetNode_v2-master\source\espArtnetNode_2.0.0_b5g\espArtnetNode_2.0.0_b5g.ino: In function 'void loop()':
C:\Users\ozdem\Desktop\ESP8266_ArtNetNode_v2-master\source\espArtnetNode_2.0.0_b5g\espArtnetNode_2.0.0_b5g.ino:261:3: error: 'doNodeReport' was not declared in this scope; did you mean 'nextNodeReport'?
  261 |   doNodeReport();
      |   ^~~~~~~~~~~~
      |   nextNodeReport
C:\Users\ozdem\Desktop\ESP8266_ArtNetNode_v2-master\source\espArtnetNode_2.0.0_b5g\ajax.ino: In function 'void ajaxHandle()':
C:\Users\ozdem\Desktop\ESP8266_ArtNetNode_v2-master\source\espArtnetNode_2.0.0_b5g\ajax.ino:28:5: error: 'serializeJson' was not declared in this scope
   28 |     serializeJson(jsonReply, reply); // JsonObject'i String'e dönüştür
      |     ^~~~~~~~~~~~~
C:\Users\ozdem\Desktop\ESP8266_ArtNetNode_v2-master\source\espArtnetNode_2.0.0_b5g\ajax.ino: At global scope:
C:\Users\ozdem\Desktop\ESP8266_ArtNetNode_v2-master\source\espArtnetNode_2.0.0_b5g\ajax.ino:46:5: error: expected unqualified-id before 'else'
   46 |   } else if (json.containsKey("success") && json["success"] == 1 && json.containsKey("page")) {
      |     ^~~~
C:\Users\ozdem\Desktop\ESP8266_ArtNetNode_v2-master\source\espArtnetNode_2.0.0_b5g\ajax.ino:59:5: error: expected unqualified-id before 'else'
   59 |   } else if (json.containsKey("success") && json.containsKey("reboot") && json["reboot"] == 1) {
      |     ^~~~

tonyxforce, 2023-09-30:
i'm having the same problems, it seems like @mtongnz abadoned this project

casesolved-co-uk, 2024-01-18:
https://github.com/mtongnz/ESP8266_ArtNetNode_v2/issues/148