mtongnz/ESP8266_ArtNetNode_v2 #148: Forget this project and use WLED!

casesolved-co-uk, 2024-01-18:
Having spent a long time updating this project and finding that it has a heap of bugs and instabilities, is difficult to develop, etc, I then stumbled upon [WLED](https://github.com/Aircoookie/WLED/).

It is far more advanced, feature rich and stable, though the DMX features aren't quite as good (no RDM).

Trust me, you won't look back!

M Tong still deserves a lot of credit for his work though. I suspect this project inspired WLED and a lot of code has been borrowed and changed beyond recognition.

Thanks!

wy1d, 2025-05-17:
> WLED is far more advanced, feature rich and stable, though the DMX features aren't quite as good (no RDM).

Cool! I'm looking through it now but not seeing support for wiring the DMX plug itself to the ESP32/8266. I actually have a bunch of ESP32 lying around I was hoping to use for an artnet adapter but also fine with picking up an ESP8266 if that's all that is supported by the project. Just having a hard time getting a parts list together for a lot that I can print a box for and use with my DMX hex pars!

casesolved-co-uk, 2025-05-17:
Yes WLED supports ESP32
You just have to convert the serial to differential serial, plenty of chips out there. ESP32 has lots of UARTs and you can assign them to any pin but check with WLED the ones it supports. The DMX functionality is pretty sketchy.
Or there's this:
https://www.racitup.com/products/supernode1

wy1d, 2025-05-17:
> The DMX functionality is pretty sketchy. Or there's this: https://www.racitup.com/products/supernode1

Hey awesome!! Thanks for the reply and link! That's exactly what I've been searching for and / or trying to build. I'm a decent enough at tinkering to build from a kit of list of components but that Supernode looks like a great option. I'm in the US though so not sure if I'll be able to get my hands on one here.

Looks like it runs WLED though so will the DMX be sketchy on it?

casesolved-co-uk, 2025-05-18:
It's actually one of my side projects. I rewrote the DMX side of things because I use it for a theatre group, but you might find the standard WLED driver does what you need. I just didn't like the look of the code because it blocks all the time it is outputting and I don't think it supports DMX input. It has been a while since I looked.

If you're feeling really brave you can try and integrate my driver into WLED, it was pretty tricky and I only tried it on the ESP8266:

https://github.com/casesolved-co-uk/DMXUART 

If you use that chip the software from the racitup.com should work on it and they are dirt cheap.

If not you can contact me through the website and I'll look into postage to the US if you're keen. Last time I purchased from the US the postage was pretty pricey. I've not tried exporting before.

wy1d, 2025-05-18:
> If not you can contact me through the website and I'll look into postage to the US if you're keen. Last time I purchased from the US the postage was pretty pricey. I've not tried exporting before.

Message sent!